RaceFlightOne/RaceFlightOne #21: Setup issue with Revolt F4 and RF1

FPVRaven, 2017-01-13:
Hi there!
I try 4 bin files now and 3 RF Configurators, but allways if it say "flash sucsessfull" and i replug my usb it will start beeping all 40 seconds and then kick me out of RF Configurator. After about 3 or 4 minutes the beeping stop and then i dont get any response from FC to my computer.
In the window where you can find "RaceFlightOne" you will find "RaceFlightRecovery" at this point and cant connect anymore. I try reinstall all i try replug usb or reopen configurator, nothing fixed it. 
Also i noticed that the blue LED on the FC start flashing if i replug the usb after sucsessfull flash.

My Setup:
- Speed Addict 210R frame 
- Emax RS 2205 2300kv Motors
- DYS XMS 30A ESCs Multishot 14.85/16kHz
- Raceflight Revolt F4 FC
- Catalyst Machineworks PDB with LC Filter (shipped with the Speed Addict frame....)
- Luminer VTX
- Graupner GR12L receiver 
- HS 1177 FPV Camera

FPVRaven, 2017-01-13:
[image]
[image]
[image]
[image]
[image]

proggod, 2017-02-17:
Hopefully this has been fixed for you

closed 2017-02-17 by proggod