RaceFlightOne/RaceFlightOne #32: S Port Telemetry VFAS - Missing

rigo3t, 2017-02-18:
Telemetry for FrSky is now working for me, but I don't get VFAS, I get only accelerometer readings and other Rx provided readings like RSSI and SWR and RxBt.

This would be the most useful thing in Telemtry, Vbat is connected on my board.

digitalomni, 2017-02-18:
what build version are you on? 185 or 189?

Enri636, 2017-02-19:
also in 190 and 191 dont work yet

rs2k, 2017-02-19:
191 should be working now. Please make sure telemetry is enabled "set telem_sport=14" and the s.port pin is connected to the USART RX pin 3 and there is voltage going to the VBAT pin on the revolt.

Enri636, 2017-02-20:
i did , but i m on V1 fc , i meen pdf wire diagram for sport , are different for V1 and V2

DebPack, 2017-02-20:
Hi the wiring diagram for Revolt V1 is wrong, you need to wire to RX3 to get telemetry.
Also VFAS working in 191

rigo3t, 2017-02-20:
On my Revolt 2 it's working also. But I didn't know the wiring was wrong in the Revolt 1. I have both boards in different quads. I will try to rewire my Revolt 1 tomorrow morning and come back with results.

Enri636, 2017-02-20:
yes work :)

proggod, 2017-02-20:
Great then i am going to close out this issue, just make sure you use RX3 on v1 or v2, we have now noted this in configurator

closed 2017-02-20 by proggod