RaceFlightOne/RaceFlightOne #37: Motors going full rpm after power off radio

Shinjo01x, 2017-02-25:
...even if disarmed before.

guys, this is dangerous!

proggod, 2017-02-25:
Need more information like which kind of radio how your fail safe is set up what version your using etc

Shinjo01x, 2017-02-25:
Radio is Taranis x9d plus. Fail safe ist set up to a Switch, disarm is set to a Switch. 

If I disarm, than power off the Radio, after 10 sec delay Motors going sull rpm for about 10 sec. 
If I disarm, than fail safe, than power off, nothing happens. 

But even if fail safe is working, this is kinda dangerous. Luckily i ll notices this with probs off. I can reproduce this every time.

Shinjo01x, 2017-02-25:
RF Version: 0.168.049_alpha_revolt_full

proggod, 2017-02-27:
Yes this bug has been fixed, always make sure your using latest when reporting bugs, give it a try in latest version.

closed 2017-02-27 by proggod