RaceFlightOne/RaceFlightOne #40: Ppm not working properly

justinquitaleg, 2017-02-26:
Cppm input doesn't work correctly on any UART ports. The "roll" axis is the only thing that works.
And also receiver autodetect doesn't auto detect ppm.

Although I don't use ppm as it has high latency, I just tested it to see if it will actually work.

rs2k, 2017-03-15:
This should be working in 199+, but the RX auto detect function does not work right now.

To setup ppm you have to skip the **RX wizard** and do this in the console.

set rx_protocol=9 if you have it on an RX pin
set rx_protocol=10 if you have it on an TX pin
set rx_usart=(whichever usart you on minus 1, so usart 3 is "2")

So to put it on RX3 TX pin you'd run:

set rx_protocol=3
set rx_usart=2
save

Then run the **RC wizard**

justinquitaleg, 2017-03-23:
:'( nope still not working. I'm using v212

justinquitaleg, 2017-03-23:
I soldered it at uart 3 rx pin.
and set it to set rx_protocol=9, set rx_usart=2

and have tried other uart ports and commands. still nothing.

proggod, 2017-03-25:
Can you give us a dump

rs2k, 2017-06-06:
PPM is now working right

closed 2017-06-06 by rs2k