RaceFlightOne/RaceFlightOne #47: Rates are lost when switching between rate modes i.e. RF Low Expo, RF High Expo and BF Rates

Rickcau, 2017-03-04:
RF Revolt Firmware version: 0.195.058 RC1
Win64 Config Tool : 0.195.0 RC1

I like the BF rate system so I was glad to see you added this in the new Win64 Config tool.

Steps to Reproduce issue 1:
1. Go to the Rates Tab and select BetaFlight Rates
2. Enter some rates using the BF style rates
3. Click Save, and a dialog will clearly show that the rates are being saved.
4. Unplug the USB
5. Reconnect the USB and navigate back to the Rates tab and not the rates that we saved are NOT read so it appears that the rates you wanted to use were not saved

Steps to reproduce issue 2:
1. Follow sets 1 - 3 for issue 1
2. So you can clear see you have set rates that are different than the default rates
3. Now, before saving, navigate to one of the RF Rates options, then back to the BF rate options
4. You will note that the rates you had entered for BF were not cached and they are resulted to the default settings

I have no idea what rates are on my board because no matter what I do they always appear as the defaults!

Arti3DPlayer, 2017-03-05:
First issue: 
Sorry, i cant reproduce this error

Second issue:
It is expected behavior, if you want to back to your values, just click "Reset values" button

proggod, 2017-03-05:
This is intentional since the rates are different scales, we want to make sure they do change or else people may try and scale them differently.   Going to close this issue, since its an intentional feature.  First one we can't reproduce so let us know if it continues.

closed 2017-03-05 by proggod