RaceFlightOne/RaceFlightOne #54: PID page save

JVBAV8, 2017-03-05:
please address PID Tuning page.
issues seen are rates values disappear, and will not save.
PID values also will not save
WIN 32 version and Windows 10

JVBAV8, 2017-03-05:
when save button is clicked, no dialog box appears confirming values were saved.
CLI dump confirms no changed values are saved.
changing values in CLI does work, but GUI PID Tuning page is unusable.

Arti3DPlayer, 2017-03-06:
Do you have 32-bit windows system ?

JVBAV8, 2017-03-06:
Yes. Win32. Tried Win64 GUI, but it won't run on my system

Arti3DPlayer, 2017-03-06:
on 32-bit is known issue, but what about 64 ? did you download 64-bit program version? and what happened ?

JVBAV8, 2017-03-06:
I downloaded it, but it won't run on my system. I tried to run it

MHessellund, 2017-03-13:
Ive seen it on 64bit version: Some times the PID/filter values doesnt load (ie. all or some of the fields are left blank, and other might load fine). I can enter values in alle field (incl. the blank ones). But when I push save, it will not save.
I can switch to another pane, and back to PID's, and every think will load fine.

rs2k, 2017-03-15:
This has been fixed in the 199+. There will be an RC3 soon. :)

closed 2017-03-15 by rs2k

JVBAV8, 2017-03-15:
Thank you. Awesome!
Releasing 199 soon?
What is RC3?
Sorry to have more questions