RaceFlightOne/RaceFlightOne #60: Flight Controller Setup Failure - Step 2

lawlessfpv, 2017-03-14:
I have a STIGG build I am working on and have installed the Skitzo FC. Trying to go through the Flight controller setup wizard and I get stuck on STEP 2. I get an error...Calibration Failed.  The board does have to be rotated on the YAW Axis 31 degrees in the frame. So this may or may not have an affect on STEP 2 when the wizard asks you to flip the quad on its nose and hit next
[image]
.  Thank you in advance for your help.
[image]

rs2k, 2017-03-15:
This feature has been added in 199+ (RC3)

Skip the first FC calibration wizard and do this in console:
set gyro_rotation=0
set board_calibrated=1
set sml_board_rot_x=0
set sml_board_rot_y=0
set sml_board_rot_z=31 (or -31 for left)
save

lawlessfpv, 2017-03-16:
Did this and went through RX Detection and Radio Setup.  But now I get this error...
[image]

lawlessfpv, 2017-03-16:
Console Command complete, but now I get all kinds of errors about RC Command timeout, RC1 command timeout, dump command errors....thoughts?
[image]
[image]

Arti3DPlayer, 2017-04-09:
@lawlessfpv i guess problem in you receiver or maybe board was out of sync. Did you try to reconnect/connect and do wizard again ?

Arti3DPlayer, 2017-04-17:
No details provided

closed 2017-04-17 by Arti3DPlayer