RaceFlightOne/RaceFlightOne #67: Taranis Current telemetry only displays integer values, not floating point

MHessellund, 2017-03-21:
The Current and Fuel sensor from the Taranis Telemetry (and Spectrum aswell) do not show the correct floating point values, it shows the integer value: Eg. 180A and 8154mAh.
As fare as I have gathered it should show 0.18A and 81.54mAh.

MHessellund, 2017-03-21:
I've done further testing, and the problem is not that floating point is missing. - The scale was so high the my Taranis did not show any peiode.
Then theres some thing else wrong. I've enteret the adc_current_factor according to the Realease notes:
1/(3,3/140)=42.4242 (I have a 3,3V adc and 140A max PDB). This factor gives 19,1A when I measure 0.21A using a multimeter.
After a shot flight, polladc showed 3749mAh. I could charge the battery with 164mAh. If I try to re-scale using the showed mAh and the mAh I could charge the battery (164/3749*42.4242), I get the following result: measured 0.3A and showing 0.8A or measured 2A and showing 1,3A.
There seems no coherency between the two results. 
I sould add that im using a proven PDB where Im use to reading current through the ADC to a micro minimim OSD /Scarab OSD

rs2k, 2017-06-06:
This should be fixed in latest versions. :)

closed 2017-06-06 by rs2k