RaceFlightOne/RaceFlightOne #81: RFC "Double Arm" should be renamed to Always Double Arm

JulianGoesPro, 2017-03-30:
there a multiple "persons" asking why after "Double Arm" they still need to double arm (first)... I know thats not going away and I think it's good but it would be wise to change the option in "Always Double Arm" so everyone gets it that disabling it will "only" change it to "not always" ;)

Thanks :)

Arti3DPlayer, 2017-04-18:
@JulianGoesPro rename "Double arm" or "Single arm" too ?

JulianGoesPro, 2017-04-21:
I think it would be best to rename both, single arm sounds too much like it would alway be that way, despite it being Double Single Arm.

If it's only gonna be that the "turn off" option, that it should be enough to rename it to "Always Double Arm" => off means Single Double arm but doesn't need to be stated (though it might be a good idea to have like a "switch" that jumps from Always Double Arm to Single Double Arm)

What do you think about that?

closed 2017-04-21 by Arti3DPlayer