RaceFlightOne/RaceFlightOne #85: implement a customizable throttle curve

X-FOOT, 2017-04-04:
why implementing a customizable throttle curve? 

because today and future builds become more &more powerfull, and a customizable throttle is realy needed.

verry light quads  with powerfull 2207 2600kv are a pain to controll at verry small & tight tracks.   here comes a throttle curve in verry handy to calm down the throttle in mid range.

a basic throttle expo wouls also be nice, to keep it simple you could disable the whole thing with a click to make use of a standard linear curve

a simple throttle curve like for axample in the old "open pilot" can make this so much better! 

a throttle curve done in the FC itself is much better, and without loosing resolution.

just giving some feedback to make the exelent RF1 even better

rs2k, 2017-04-21:
This has been added as of build 233. :)

closed 2017-04-21 by rs2k

X-FOOT, 2017-04-26:
great, gonna check it.  greets Hans from X-foot