RaceFlightOne/RaceFlightOne #90: CLI commands have no effect on vTx channel

JulianGoesPro, 2017-04-15:
Example... 

`#ms Unknown Command:...vtxinfo
#me vtx.vtxDevice=2
#me vtx.vtxBand=4
#me vtx.vtxChannel=5
#me vtx.vtxBandChannel=37
#me vtx.vtxPower=1
#me vtx.vtxPit=0
#me vtx.vtxRegion=0
#me vtx.vtxFrequency=5843
#ms Unknown Command:....going
#me vtx.vtxDevice=2
#me vtx.vtxBand=4
#me vtx.vtxChannel=0
#me vtx.vtxBandChannel=32
#me vtx.vtxPower=1
#me vtx.vtxPit=0
#me vtx.vtxRegion=0
#me vtx.vtxFrequency=5658
#ms Unknown Command:...doing
#me vtx.vtxDevice=2
#me vtx.vtxBand=4
#me vtx.vtxChannel=5
#me vtx.vtxBandChannel=37
#me vtx.vtxPower=1
#me vtx.vtxPit=0
#me vtx.vtxRegion=0
#me vtx.vtxFrequency=5843`

the vTx seems to respond with the "suggested" channel but checking if it actually moved (vtxinfo) shows it didn't... looking at my goggles band scanner... same story... vtx isn't changing channel at all :(

GateslayerFPV, 2017-04-19:
I couldn't even get VTXON and VTXPIT to work. VTXINFO worked fine.

rs2k, 2017-06-06:
fixed in recent buids

closed 2017-06-06 by rs2k