RaceFlightOne/RaceFlightOne #91: Feature Request: Add low raceband frequencies for TBS Unify Race

hjns, 2017-04-16:
Request from slack user:

Add low raceband frequencies for TBS Unify Race to the telemetry script.

cturn, 2017-04-20:
I second this request :dancer:

rs2k, 2017-04-21:
What are the low race band frequencies?

cturn, 2017-04-21:
Low Band Frequencies are:
5325mhz
5348mhz
5366mhz
5384mhz
5402mhz
5420mhz
5438mhz
5456mhz

hjns, 2017-04-21:
I'm wondering if there are multiple ones, I found this on the web;

[image]

cturn, 2017-04-21:
I think they're are, was wondering that as I was typing the list as they didn't sound right.  But, the ones you listed are obviously the correct unify L-band channels, thanks!

rs2k, 2017-06-06:
Planned for release 1.0

PGIII, 2018-03-19:
added in latest

closed 2018-03-19 by PGIII