RaceFlightOne/RaceFlightOne #96: Frsky s.port sensor lost warning

ankjaers, 2017-04-25:
There seems to be a bug that causes my taranis to give a "sensor lost" warning very often. If I flash betaflight and enable telemetry this doesnt happen. It happens happens when the quad is in the air. But seems to happen more frequently when the remote is close to the quad. So it's likely not a bug. But it's surely something thats super annoying.

deepimpactio, 2017-04-25:
Kalyn mentioned it would be adjusted in future builds. Appears packets are
being sent to fast for S.Port.

  Screenshot_20170425-062128.jpg
<https://drive.google.com/file/d/0ByVtmlwafXpKbDN0QUtyWmFnLVk/view?usp=drivesdk>

<https://github.com/RaceFlightOne/RaceFlightOne/issues/96>, or mute the
thread

rs2k, 2017-06-06:
fixed in latest builds

closed 2017-06-06 by rs2k