RaceFlightOne/RaceFlightOne #99: Add blheliS RGB Led support (change colours)

vierpunktzwo, 2017-05-01:
As there is no passthrough available, and I would like to use the LED feature, it would be nice to have a simple checkbox in GUI that allows to turn on each of the 3 led on the ESC PCB like ipeaka has (for example).

rs2k, 2017-06-06:
Checking into this now.

PGIII, 2017-06-12:
Blheli Firmware should handle the leds on its own, it changes them as its starting up and then makes them green when theyre started up and have signal, unless it differs from esc to esc it should just work when theyre updated

vierpunktzwo, 2017-06-12:
Not really preston, In fact there are some esc with a free defineable rgb LED. Check the newest Suites and you will see a option for those. It's not a normal status led, it's a feature to replace additional Led's ;)