RaceFlightOne/RaceFlightOne #102: ISSUE: 0.246.00 RF1 Configurator

chargerdude70, 2017-05-03:
Win64 Setup >> 4. Update ESC: After step 1 is complete and click "Next", it bootloops back to step 1.

Arti3DPlayer, 2017-05-03:
@chargerdude70 `Remove your props !!!` should be your first step, then when you click next what happen ?(it should show you window with loading progress)

chargerdude70, 2017-05-03:
This credit card comes up. 
[image]

After it closes it reverts back to this. 

[image]

It never moves onto next step. Just loops to first step. There is no "Finish" or "Done" option

Arti3DPlayer, 2017-05-04:
@chargerdude70 I tested on windows x64 it works for me. I will try to ask someone to test too

rs2k, 2017-06-06:
I'm closing this issue. Please reopen if it's still happening. :)

closed 2017-06-06 by rs2k