RaceFlightOne/RaceFlightOne #114: Configurator: People seem to miss the "make sure the battery is disconnected" in motor wiz

hjns, 2017-05-26:
A lot of people seem to miss the "make sure the battery is disconnected" in motor wiz and get the "Make sure your drone is immovable" error.  I suggest making the DISCONNECTED larger and more prominent and  adding something like "This is also caused by the battery not being disconnected in the previous screen" to the "Make sure your drone is immovable" error popup.

Arti3DPlayer, 2017-05-26:
could you give a screenshot please

hjns, 2017-05-26:
[image]
[image]

rs2k, 2017-06-06:
[image]

This should be change to "Please make sure your battery is DISCONNECTED and your quad is completely still."

Arti3DPlayer, 2017-06-06:
fixed

closed 2017-06-06 by Arti3DPlayer