RaceFlightOne/RaceFlightOne #122: Taranis "sensor lost" warning

carloscampos775, 2017-06-17:
Using OpenTX 2.2 the Taranis will sound intermittent warnings about a "Sensor Lost". After going through the sensors, I found the issue is with the VFAS sensor. The VFAS sensor will stop updating at random intervals.

rs2k, 2017-06-17:
This is fixed in soon to be released builds. :)

carloscampos775, 2017-06-17:
Still an issue with 295

carloscampos775, 2017-07-18:
Still have the issue with 301 and an updated XSR

MilanFPV, 2017-08-04:
I have the same issue. On 306 with updated XSR.

_Sent from my Motorola Moto G 2014 LTE using [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_

rs2k, 2017-08-13:
Does 313 fix it? My taranis never does this.

MilanFPV, 2017-08-14:
Nope, in 313 its still there.

_Sent from my Motorola Moto G 2014 LTE using [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_

carloscampos775, 2017-08-15:
I can also confirm that the problem still there. Installing 313 helped.
However, the VBAT reading is very slow to update. 313 helped to reduce the
warning to about once per pack

arjxbh, 2017-10-11:
Having the same issue here, although it seems less frequent after updating to 320.

darrindevine, 2017-10-14:
It sounds like this is a known issue with FrSky firmware effectively receiving telemetry from newer FCs.  

Firmware:
https://www.frsky-rc.com/xsr/
"Improve the compatibility between the S.Port and F3 FC."

Video:
"Upgrading FrSky XSR Firmware. And fixing telemetry with the XSR after telemetry stopped working"
https://www.youtube.com/watch?v=VCpzlijHNVw

cturn, 2017-10-14:
Good find, wasn't aware of any "recent" XSR firmware updates, I'll be checking/updating firmware on all my RX's now.  I have this same issue on a couple builds, just disabled telem, was too annoying.  Sounds like this has been fixed in newer RF1 builds?  But curious if this RX firmware will resolve it on my older RF1 quads.

darrindevine, 2017-10-14:
Well I'm on RF1 RC17(newer RC and betas available) and just tried the newest XSR firmware. It didn't change the Sensor Lost issue. Maybe FC firmware needs updating as well the RX. I won't be trying that today.

PGIII, 2017-12-12:
should be fixed in 420, we can look into it some more if its not

closed 2017-12-12 by PGIII