RaceFlightOne/RaceFlightOne #126: Configurator: linux 295 is missing the .desktop file and icon

hjns, 2017-06-18:
The 295 configurator doesn't have the .desktop file that makes it easier for people to add the app to the desktop or desktop menus.   The desktop icon files is also missing.

Arti3DPlayer, 2017-06-18:
@hjns Ok, i will add .decktop file. But what do you mean > The desktop icon files is also missing.

Arti3DPlayer, 2017-06-18:
ah, i understand

hjns, 2017-06-18:
This is the icon that the .desktop file used, it was at the top level in the zip file, the same place the executable file is.   It's not in 295.  The old .desktop file referenced this icon.  It didn't work unless the user fixed the path in the .desktop file but it was there if they wanted to use it.  It can't work by default because the .desktop file wants a fully qualified path but at least users who knew linux could change the path in the .desktop file to match their installation.

[image]

Arti3DPlayer, 2017-06-19:
@hjns i'm not very familiar with windows, what in `.desktop` file should be ? do you know ? As i tested i can create Desktop shortcut(without icon for now)

hjns, 2017-06-19:
I haven't seen any complaints about windows.  Windows users should be able to just right click on the exe and create a shortcut for the desktop.

For linux this one should work, it's a modified version of the one that was in earlier releases. Put it in a file named  RaceFlight.desktop the same place as the executable is.  I've attached a zipped copy in case github mangles the text.  Put the png above in the same place.  The path for the icon will be wrong but knowledgeable linux users will change it and it won't hurt anything for others.  It's one of those stupid things where the icon has to have a full path or be installed in a particular directory reserved for icons. The version should always stay 1.0, it's the version of the desktop file specification, not the program version. 

[Desktop Entry]
Version=1.0
Name=RaceFlight
Comment=Crossplatform configuration tool for RaceFlight flight control system
Exec=bash -c "\\$(dirname \\$1)/RaceFlight" x %k
Icon=/home/hwh/rf/rf1/configurator/linux-x64-295.3/rf_icon_128.png
Terminal=false
Type=Application
Categories=Utility;

[RaceFlight.desktop.zip](https://github.com/RaceFlightOne/RaceFlightOne/files/1085709/RaceFlight.desktop.zip)

Arti3DPlayer, 2017-07-11:
@hjns I guess i will implement this when we build some Installer for configurator to handle this job

hjns, 2017-07-11:
If you build an installer it should probably do the udev rules as well.  They're documented in the pdf guide on the RF download page.

closed 2018-11-29 by PGIII