RaceFlightOne/RaceFlightOne #127: Motor Scores page layout request

Matthias7Song, 2017-06-18:
Motor scores result page layout is like this;
Motor 1                  Motor 3
Motor 2                  Motor 4

But it would be easier for users to find out which motor is which if there is a copter image such as we can see motor rotation setup page in the center of motor scores result page and put the result values aside to each motor. 

Or if the layout can be changed according to RF1 motor position like this;
Motor 1                  Motor 2
Motor 4                 Motor 3

Many users including me motor mixing and do not care about which ESC wire is put to which pin on Revolt. So it is a little bit confusing to find out which motor is bad when Motor 1 result is bad.

PGIII, 2017-06-26:
@Arti3DPlayer

Arti3DPlayer, 2017-06-28:
Will fix in 0.300.0+ version

closed 2017-06-28 by Arti3DPlayer