RaceFlightOne/RaceFlightOne #133: Feature request: Assign Airmode to a switch

rcquad808, 2017-06-19:
Program the ability to activate and deactivate Airmode on a switch.

PGIII, 2017-06-21:
There shouldn't be any reason to not always have control why do you want to be able to turn it off? we don't use airmode its a different method that shouldn't cause any issues

rcquad808, 2017-06-21:
I use Airmode all my racing or just flying around, but i would like to turn it off during landing cause it make it so much easier  and controlled. Raceflight use to have the landing feature but now it gone. For safety, being a zero throttle/ quad on the ground the quad should disarm (to avoid accidental throttling back up if the user fails to manually disarm the quad) as the non-spinning blades would no longer be a sure sign that the quad is disarmed.

closed 2018-03-19 by PGIII