RaceFlightOne/RaceFlightOne #134: Log playback timing fast

apocolipse, 2017-06-19:
### Summary
The playback speed of blackbox logs is faster than it should be

### Steps To reproduce
Load blackbox log with accompanying video.
Sync video to ARM start
Playback log

### Expected results
Expect the log to play back in sync with recorded video

### Actual results
After about 15 seconds of playback, the log is ~1s ahead, by the end it is several seconds ahead.

### Why this matters?
When attempting to troubleshoot issues in flight it is often good to have visual/audible input to compare directly to that time in logs, current playback time for Blackbox logs being quicker than video either requires video editing for time or constant re-syncing of video to logs to attempt to match expected flight issues.

PGIII, 2017-06-21:
Which version are you using ?

apocolipse, 2017-06-21:
Happens with all the latest beta test builds, tried using both Cleanflight and Betaflight blackbox viewers. I haven't tried with 29x builds yet, I took some footage and logs today on 299 I'll test and report back.

rs2k, 2017-06-22:
Hmm, that's an odd ratio that doesn't seem firmware related. I think the next thing to try is to arm the quad for a known time and see if the arm time in the log is the same... Say 30 seconds to see if the numbers match.

apocolipse, 2017-06-23:
@rs2k after much trial and tribulation trying to do this test (OSD is conflicting w/ logging), I ran the test, synced my 2nd arm switch flip w/ starting timer on my phone, ran for 30.11 seconds, blackbox log shows 28.6 seconds, so its about 1.4s fast over 30s.

cr3pitus, 2017-06-28:
I have the exact same results and have seen others post about it in slack.  Has been an issue for me on every version I've tried, including all the 29x versions.

apocolipse, 2017-08-07:
Update:  Still seeing this on 300+ builds (300 and 306 tested) on all quads.

rs2k, 2017-08-13:
The logging is not scheduled. It's based on gyro's output. If the frequency is off slightly I would expect this to happen. I'll change the log to use the cpu to decide time instead of iterations.

closed 2018-03-19 by PGIII