RaceFlightOne/RaceFlightOne #138: Remove long beeper tone on plugin

mpr52, 2017-06-24:
On 295 the beeper goes for what seems like forever on plug in. Can I stop this Somehow?

PGIII, 2017-06-26:
it should keep beeping , are you sure you have the buzzer hooked up right and the fc is working properly

MilanFPV, 2017-07-26:
I had the same problem. Its because you activated smart audio and you have connected the tramp vtx. Its a bug, disable smartaudio and it will beep normaly when you plug in the battery.

rs2k, 2017-08-13:
Yes, this is a bug. The buzzer routine doesn't run while the vtx is being searched for.

robomaniac, 2017-10-22:
is this what the buzzer do?
https://raceflight.slack.com/files/U1NAGQ4RF/F76B09RTM/buzzer_at_start_up_is_weird.mov

I set the VTX device to "other" and does not work, When I plug USB I hear the buzzer beep.

Version: 0.320.1
FC Version: 0.320.125 BETA

closed 2018-11-29 by PGIII