RaceFlightOne/RaceFlightOne #140: Making FC sound buzzer when battery is low

FeBoCaMoAl, 2017-06-24:
Hi. Would it be possible to add to your FCs' firmware a feature to have it sound the buzzer when the battery is low? Since it already gets that info, I would think it could be programmed into the software.

This feature would be great for both FPV pilots using cameras without OSD (like my 2" Millivolt build), and for line of sight.

Thanks! I appreciate your time.

PGIII, 2017-06-26:
its been in there for a while now, if its not working we can look into but weve had a low voltage alarm since before rc17

PGIII, 2017-06-26:
already another issue for quiet buzzer, sounds like its working just isn't very loud, we can look into it

closed 2017-06-26 by PGIII

FeBoCaMoAl, 2017-06-26:
Oh, ok. One of your reps said this feature didn't exist. My millivolt is
still on the way.