RaceFlightOne/RaceFlightOne #141: Firmware update and motor spin issues

Rafyel, 2017-06-26:
Firmware update and motor spin issues with 4in1 ESC's like the DYS F30a, and now the xRotor 30a 4in1 Mini where the 4th motor will not spin after updating the firmware through the configurator. This has been a pain in the you know what for over a month I can't get my quad in the air. I've tried three (3) DYS 30a 4in1 ESC, and now HobbyWing xRotor 30a 4in1 and they all have the same issue. Motor #4 doesn't spin after the RaceFlight Client attempts to update the ESC's. After doing so I can't even use BlHeli Suite to recognize the ESC's to get the ESC's working again. Argh!

However, the Spedix 30a 4in1 works just fine!

PGIII, 2017-06-26:
can you upload a 1wire read from both escs also what verison are you using

Rafyel, 2017-06-26:
Here is the 1wire Read log:

[1wire_read.pdf](https://github.com/RaceFlightOne/RaceFlightOne/files/1103599/1wire_read.pdf)

Rafyel, 2017-06-26:
XRotor Micro 40A 4in1 BLHeli-S DShot600
http://www.hobbywing.com/goods.php?id=588

Rafyel, 2017-06-26:
It comes with A_H_50 firmware, BlHeli_S v16.5.

[image]

PGIII, 2017-06-28:
It says they have A_h_25 on them, I don't think we have either A_H_50 or 25 in there so it can't flash the escs, and it looks like it's on 16.51 which isn't the version 299 flashes, it looks like this is an issue with the escs them selves

Rafyel, 2017-06-28:
You do have A_H_25 and 50. See the screenshot from the manual. Position 36 and 39. They worked out of the box. After attempting to flash, motor 03 is now Unreadable.

PGIII, 2017-07-07:
the 1wire read you posted has 16.51 on it which 299 does not flash, it also doesnt say theres an upgrade available which means it doesnt have the hex for that esc, so it wouldn't even attempt to flash the escs when you run the 4th wizard, like i said this seems to be an issue with the escs since all RF1 would do is read the escs, you can type 1wire list in the console to see a list of all the hexes we have in there

proggod, 2017-07-24:
This seems more of a tech support issue, so going to close it as a github issue.  Please contact someone on slack and make sure you are running the latest RF1.

closed 2017-07-24 by proggod