RaceFlightOne/RaceFlightOne #142: VFAS is not updating properly, causing intermittent telemetry

hebster, 2017-06-26:
Using 299 and earlier, the VFAS telemetry from an XSR (tested both firmware 170605 and 151118) is intermittent, causing "sensor lost" as well as delayed VFAS values sent to X9D plus.

mpr52, 2017-06-26:
Also experiencing this. On rc17 and 295

GiulianoDeMicheli, 2017-06-29:
Also experiencing this. On 299. Taranis Plus and XSR EU-LBT on TX1 not updated to the last fw and Tramp HV connected to TX1

tjphippen, 2017-07-12:
I'm getting this too. Haven't tried yet but heard updating the XSR firmware to the beta version fixed this.

papayou66, 2017-07-25:
It will not work @tjphippen i tried yesterday to update my xsr

GiulianoDeMicheli, 2017-09-14:
the problem has come back with RaceFlight One BETA Build 0.320.125. 
My config: Taranis Plus and XSR EU-LBT on TX updated to the last fw, Tramp HV connected to TX4 inverted, revolt v2

PGIII, 2017-12-12:
should be fixed in 420

closed 2017-12-12 by PGIII