RaceFlightOne/RaceFlightOne #146: Buzzer it's on during motors spinning

GiulianoDeMicheli, 2017-06-30:
Buzzer it's on during motors spinning. I've got buzzer on a switch with arm. Revolt 2 with Taranis Plus and XSR. Fw beta 299. No buzz issue with 238.
Thx for your work :+1:

PGIII, 2017-07-07:
Do you have the buzzer on the same switch as arm? can you upload a picture of the configuration page with all the modes

GiulianoDeMicheli, 2017-07-10:
Thank you for the reply. I will upload the image as soon as possible. Yes I've got buzzer on the middle position of the arm switch and arm on the top position.

MilanFPV, 2017-08-04:
Thats why the buzzer is set up to beep when lipo cell is under 3.8V. Type in console "dump" than it will show you two commands for the buzzer to Turn of or regulate the voltage.

I think 3.8V is not good for stock setting.

_Sent from my Motorola Moto G 2014 LTE using [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_

rs2k, 2017-08-13:
Does this problem still exist in 313+?

closed 2017-11-21 by proggod