RaceFlightOne/RaceFlightOne #149: Death SPin/ Infinite Yaw spin

RoninMedia12, 2017-07-04:
On 299 still get Death spin if hit a gate or object!   seems to spin out for about 20-30secs then will correct itself if you are in open space and don't disarm.

PGIII, 2017-07-05:
We've hit quite a few gates and haven't had any issues, it sounds like you may have an esc going out or a prop is getting stuffed

RoninMedia12, 2017-07-05:
This has happened on 3 different quads and seems to happen more often on my
lighter builds.  It is strange it will recover some times after spinning
for 30 plus seconds.

RoninMedia12, 2017-07-05:
I know I have Dvr of at least on instance where this occurred and ended up in the trees as local field had it happened again yesterday and No bent props pulled the quad out of the trees and it flew immediately. No logs I don't think But can try to capture some of when it happens.   I was at a race on Sunday and had to disarm it was witnessed by 30+ people they watched me pick up the quad after and test hover it that was on my 4" build.

PGIII, 2017-07-05:
its possible the spin stopper may need to kick in sooner for 4in, we use 3in and 5in and dont have any issues, we can make the spin stopper adjustable in the next release so you can adjust it

RoninMedia12, 2017-07-05:
Thank you thay would be a good thing to be able to adjust for different
builds as my 5 inch does it too but does not seem to be as much. But does
still occur.

RoninMedia12, 2017-07-09:
So can you please explain the ranges for the spin stop in 300.112? and commands   I would like to test it out some today as I have the day off and the weather is nice out.   Thank you

PeteG-FPV, 2017-07-11:
I've had the yaw spin of death on my quads too (7 different race quads). All 5", running 299. Different ESCs and props.

proggod, 2017-07-24:
There is a facebook post on how to adjust this, you can change it in the cli.  We also recommend running reverse prop rotation

closed 2017-07-24 by proggod