RaceFlightOne/RaceFlightOne #157: Feature Request: modification to dump in cli

justingoesfpv, 2017-07-09:
Modify the dump command to accept a parameter to allow for the dumping of PIDS without being in the percentages. This would allow users like myself to easier track numerical values versus the percentage values between builds(to use percentages I have to track the "max" which isn't dumped in the cli either for example 432.9 for roll p). I currently keep dumps from each default config to compare between builds for troubleshooting issues like hot motors etc. Having PIDS in numerics would help simplify the process.

Arti3DPlayer, 2017-07-09:
@justingoesfpv Use Show pids in percent option on PID tuning tab to see values

justingoesfpv, 2017-07-10:
Artem,

I know I can visually see them in a locked edit box, and from there I can
note them down. From there I can also calculate the hard coded PID #s that
the percentage system is working from. I can and currently do do this and
track changes between beta builds which helps identify tune changes in the
builds, so as a public beta tester I know what I can expect a bit better.

But what i can tell from reading, and based on reintroducing the ability
too see the numbers versus percentages, more than just me is interested in
this data. So simplifying the process of collecting and storing by
modifying the dump command to accept a 1 as a second parameter dumping the
numerics just seemed basic.

Then again it would probably behoove the whole system to make the
percentages fully optional, I've tested decimal precision and it seems to
work for fine tuning with the percentages, but based on percents you will
only ever get so close. But im guessing that doesnt go well with the tx PID
tuning abilities, as personally is just assume shoot myself in the foot the
change PID for RPY by 20pts each via the current LUA script.

PGIII, 2017-07-11:
The percentage system is what we're moving to completely and what we want everyone to get used to, you can see what the pids are in the GUI, but there are many reasons why we moved to percentages, one of the biggest ones is to make tuning easier, the percentage based means pids should more easily transfer from version as version as well which should make it so you don't have to do as much tuning when you update

closed 2017-07-11 by PGIII