RaceFlightOne/RaceFlightOne #158: RFOSD Voltage Bug (rare)

JulianGoesPro, 2017-07-11:
Hey, I noticed my voltage sometimes shows 15 5 instead of (I assume) 15 50, not sure if that is a rounding bug or something, it looks like it is only appearing with 15.50volts, other then that it shows two decimals (which I like) :)

BTW are there going to be "." added so it shows like 16.79 instead of 16 79? (not sure what I like more but just curious ;) )

Thanks!
[image]

apocolipse, 2017-08-07:
Same here, it's reporting `15 5` when it's at `15.05`, so needs left padding (and a decimal point)

rs2k, 2017-08-13:
This will be fixed before release. The current OSD was just a simple implementation to get the important code in place.

closed 2018-03-19 by proggod