RaceFlightOne/RaceFlightOne #159: LED addressing issues

friszky, 2017-07-11:
It seems like the first LED on a strip will not respond, or act randomly. The others work totally fine and respond to inputs. 

LED one will either be off, or a random steady color, but *does* change color at random intervals.

This is on a Revolt V2, using Adafruit 8x RGB LED strip on PCB. Other strips exhibit same behavior. Changing the "LED Count" does not help.

PGIII, 2017-07-27:
What version are you on ? and did it work on previous verisons ?

friszky, 2017-07-27:
I'm now on rc18, previously on 299, and before that RC17. It didn't work on
any of them as far as i remember.

PGIII, 2017-07-30:
I can recheck rc17 I'm pretty certain it worked then, I can replicate it on the later builds, I'll have to take a look

charmlasse, 2017-08-06:
I have Revolt version 1 and other type of LED-board and I have noticed exactly the same issue. Worked fine with revolt195.bin, but not with revolt229.bin or revolt300.bin.

proggod, 2017-11-21:
should be fixed in 402

closed 2017-11-21 by proggod

charmlasse, 2017-11-22:
Works fine now. Thanks!

Regards
Niklas

Skickat från min Samsung Galaxy-smartphone.