RaceFlightOne/RaceFlightOne #160: Feature Request: Custom Modes

tjphippen, 2017-07-12:
I'd like the ability to directly assign radio switch positions/modes to trigger console commands. While this feature could be used in many ways my initial use would be to have my Custom Mode 1 trigger "sach 30" to change my VTX to channel F7. 

This could be as simple as a text box on the configuration page for each custom mode but could become even more beneficial if variables were allowed & small formulas were permitted as well. E.G. "set sach (sach + 1)" which would simply increase the current channel by 1. I could even create "siforek mode" that runs various commands to set my rates, filters etc. & share it with others..

Again I'm sure there's a million other ways this could be used. It would open up the possibility for users to add modes that you may already have planned sooner & even contribute things you may want to later add to the current included modes.

tjphippen, 2017-07-12:
This could also be used similarly to the CF in flight adjustments modes.

proggod, 2018-03-19:
put this for v1.1