RaceFlightOne/RaceFlightOne #161: TRAMP HV TELEMETRY

GiulianoDeMicheli, 2017-07-13:
Tramp HV telemetry via lua script with Taranis Plus does not work on tx3 or another tx pin. Please solve it. Thx for RC18!

spacebubi, 2017-07-13:
i've same problem...i've soldered on UARTrx3 the s.port of X4RSB frsky and from TRAMP HV the T wire on the first time at UARTtx3, but i read can give some problems..than i swapped the T wire into another TXuart free on RevoltV2...but...nothing to do. Can resolve it? or better make a detailed guide for all if there a solution? thanks

spacebubi, 2017-07-13:
there's some specials commnads in console to put ? the smart audio of TBS in tab configuration..is equal to TRAMP HV  "T" wire??? help us please

SkeemoFPV, 2017-07-16:
It worked on one of the earlier builds (266), but it hasn’t worked since. I know it not my set up as I can flash 266 back on the Revolt v2 and it works, but nothing on any other build.

CarbonV, 2017-07-17:
I have the same problem, following

Quickred, 2017-07-17:
Same problem for me.

Lua script on Taranis X9D+ didn't detect TrampHV with 299
connect S.Port from X4R-II to RX3
telemetry works
connect Tramp T on TX4
bridge the pad for TX4 to NOR
in CLI set telem_smartaudio = 0 
in CLI set telem_tramp = 4

no VTX detected :(

with or without NFC connected makes no difference

papayou66, 2017-07-25:
Same issue on Tx3 or Tx4

MilanFPV, 2017-07-26:
Same problem here!!

hjns, 2017-08-03:
I haven't got a tramp to test but it's been reported that like some other variables telem_tramp is zero based instead of one based and that for tx4 it's set telem_tramp=3   A user says this works for him:
[image]

Note: this would make it impossible to use tx1 for tramp since 0 is disable.

MilanFPV, 2017-08-04:
@hjns thats my facebook post, thanks for sharing here. I figured it out with a guy in a german group.

I only tried it on tx4 but maybe it also works on other uarts..

_Sent from my Motorola Moto G 2014 LTE using [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_

MilanFPV, 2017-08-04:
Here's the complete post:

------ !!! TRAMP HV DETECTION !!! ------

Today I figured out how to conect the Tramp to change VTX settings on the LUA Script.

Conect the telemetrie of the Tramp to TX4 (jumper normal) and then type this 3 commands in the console:

"set telem_smartaudio=0" - "set telem_tramp=3" - "save"

I dont know why, but only so does it work!!

_Sent from my Motorola Moto G 2014 LTE using [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_

GiulianoDeMicheli, 2017-08-04:
Please advise that it only works with TRAMP V2!

rs2k, 2017-08-13:
This should be fixed in 313+ 
Please reopen if this issue is not completely resolved.

closed 2017-08-13 by rs2k