RaceFlightOne/RaceFlightOne #165: Adc for bolt/telemetry issues

jrhoad, 2017-07-14:
On 299 I had set ADC current factor to 34.2 as noted in previous release notes, mah consumed seemed fairly accurate with the bolt ESC, although my DX6 would not display a current reading. Upon flashing to 300, I set ADC to 34.2 once again, and the mah consumed seems to be underestimated by a factor of ~20x (reads 40mah consumed when I'm closer to 800mah), as well as still not displaying a current reading. Not entirely sure if this is a setup, config, software, or user error.

Revolt V2
Bolt
35v 1000uf caps
Spektrum telemetry rx
Dx6
Flashed with 300
Gui also 300

jrhoad, 2017-07-14:
Also noteworthy, CLI displays set ADC current factor to 3 after I input set ADC current factor 34.2

hjns, 2017-08-01:
That last comment is an example of user error.   Instead of typing it the way the dump shows you added spaces.

hjns, 2017-08-01:
This is a duplicate of issue #147

jrhoad, 2017-08-02:
I both typed it exactly as shown and copy/paste it with the same result Everytime. Even attempting to use other numbers yields adc current factor 3

hjns, 2017-08-03:
You need to find some way someone other than you can duplicate this.  Anyone looking at this is going to do the same thing I did, type it in right and have it work.  People change this value all the time and no one has the problem you have unless they mistype it.  Below is the right way and what you get with various entry errors.

--- right ---
set adc_current_factor=34.200
#me adc_current_factor=34.200

--- comma instead of period ---
set adc_current_factor=34,200
#me adc_current_factor=34200

--- space after equal sign ---
set adc_current_factor= 34.200
#me adc_current_factor=34.

--- space before equal sign ---
set adc_current_factor =34.200
#me adc_current_factor=34

--- space before and after equal sign ---
set adc_current_factor = 34.200
#me adc_current_factor=3

jrhoad, 2017-08-03:
And there it is! Thank you for your help Hank, I will try this when I get home.

rs2k, 2017-08-13:
This should be added to the gui.

proggod, 2017-11-21:
should be fixed in 402

closed 2017-11-21 by proggod