RaceFlightOne/RaceFlightOne #169: Drift to the right 300+

dronedout, 2017-07-17:
0n 300 I notice that mid flight I start getting a drift to the right. When I start on lift off it hovers stable and does not have it but if I fly and do a lot of hard turns then I get the weird drifting starting mid flight and I have to compensated by fighting it with the sticks to the point where I let go the quad will just lean of veer off to the right. All mounting is stable and nothing is on my gyros. This is happening on my v2 build as well as a f400 build and all mounting and stable and level.

led3475, 2017-07-17:
I'm having a very similar issue after going from 299 to 301, except I have a pitch and roll drift. I also know another that has a yaw drift on 301 as well. ANY idea's?

2mod, 2017-07-17:
id assume that what you guys are having is more of a tuning issue, rather than a fw problem - try increasing I on the axes on which your experiencing the drift and it should go away, should you experience an emerging bounce back after quick flips/rolls/spins youve gone too high on I

PGIII, 2017-07-19:
Post logs from the quads that are having issues

carloscampos775, 2017-07-21:
I can also confirm this issue with 301. Noticed yaw issues as well.

closed 2017-07-23 by rs2k