RaceFlightOne/RaceFlightOne #171: GUI 307 or 306.116 FW issue

chargerdude70, 2017-07-24:
When changing Rates in "Profile 1" to very tame rates, I get a straight line in the graph and 1deg/sec  roll for all axis.

[image]

TomT4S, 2017-08-01:
I can reproduce this issue - 

It appears when saving different profiles, the rates drop down box reverts to "Rates" which seems to be a null value. When re-choosing an option in the drop down, the rates will work. 

Also noticed smoothing reverts to a "0" value, which could cause full throttle on arm, even though the GUI does not allow the smoothing to be 0, once unplugged and re-plugged in the "0" value is retained. 

I have not tested this real world though.

Arti3DPlayer, 2017-08-01:
@TomT4S +1 seems it is board save issue @rs2k

TomT4S, 2017-08-02:
Fixed in 309.

closed 2017-08-06 by Arti3DPlayer