RaceFlightOne/RaceFlightOne #172: GUI 307 or 306.116 FW issue

raceflightkid, 2017-07-24:
come, guys, what going on, build 306 and GUI 307  1wire read/any thing to do with ESC is not working for me GUI reboot with hid the error

Arti3DPlayer, 2017-07-24:
@raceflightkid screen please

raceflightkid, 2017-07-24:
[image]

raceflightkid, 2017-07-24:
i leave it on 1 all is ok , but on 33 no radio inputs i allways use 1 any way

Arti3DPlayer, 2017-07-24:
@raceflightkid I don't see issue. What did you mean: > no radio inputs  ?

raceflightkid, 2017-07-24:
well i put rc smoothing on 33  and arm the quad theres no radio inputs , no throttle/yaw/roll/pitch

papayou66, 2017-07-25:
I just checked i do not have this issue

raceflightkid, 2017-07-25:
ok im going to close it , it as to me my end

closed 2017-07-25 by raceflightkid