RaceFlightOne/RaceFlightOne #173: 306 profile bug

MilanFPV, 2017-07-26:
On firmware 306 the profiles still have bugs.

PGIII, 2017-08-02:
can you be more specific?

MilanFPV, 2017-08-02:
When I rename the profile1 / profile 2 / profile3 and hit save and than reconect the FC the profiels are no more there. All Rates and and RC Smothing are deleted. Also when the name of the profile is for example "Skitzo Rates" after I reconected the board, the name changed to "SkitzoR"

Sorry for my english, but try to reproduce the problem.

Arti3DPlayer, 2017-08-03:
@MilanFPV https://github.com/RaceFlightOne/RaceFlightOne/issues/171  is this what you mean?

MilanFPV, 2017-08-04:
@Arti3DPlayer no that isnt what I mean

_Sent from my Motorola Moto G 2014 LTE using [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_

rs2k, 2017-08-13:
Fixed in most recent

closed 2017-08-13 by rs2k