RaceFlightOne/RaceFlightOne #174: firmware: console data entry needs a more robust parser

hjns, 2017-08-03:
The console number parsing is very narrow minded, common errors like adding spaces cause the numbers to be misread but don't show an error.

For example:

--- right ---
set adc_current_factor=34.200
#me adc_current_factor=34.200

--- comma instead of period ---
set adc_current_factor=34,200
#me adc_current_factor=34200

--- space after equal sign ---
set adc_current_factor= 34.200
#me adc_current_factor=34.

--- space before equal sign ---
set adc_current_factor =34.200
#me adc_current_factor=34

--- space before and after equal sign ---
set adc_current_factor = 34.200
#me adc_current_factor=3

MilanFPV, 2017-08-04:
Thats normal. When you hit space, the command is over.

_Sent from my Motorola Moto G 2014 LTE using [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_

hjns, 2017-08-04:
It's not normal, just the way it is. That's the reason for the suggestion that it be improved.  If it took it as command over and showed an error because it hadn't gotten a number it would be better.  It doesn't see the space as command over because it continues but mis-reads the number.

rs2k, 2017-08-13:
This is very low priority. The GUI is are main configuration point. Minor issues like this will be fixed after release 1.0

closed 2018-11-29 by PGIII