RaceFlightOne/RaceFlightOne #176: DropDownMenus for Tramp and OSD

JulianGoesPro, 2017-08-06:
Just like SmartAudio has it's DropDownMenu could we have one (right next to it) for Tramp TLM and OSD Tx selection as well?

Now
[image]
New?
[image]

Arti3DPlayer, 2017-08-06:
@JulianGoesPro You suggest to change switch widget to dropdown widget ?

JulianGoesPro, 2017-08-06:
add one for TrampTLM (which corresponds to set telem_tramp=X) and change the switch to dropdown for the OSD (which does not only enable it but lets you select the Tx like the SmartAudio dropdown lets you => set telem_osd=X)

hjns, 2017-08-07:
Since tramp and SA are never enabled at the same time I'd just add the tramp settings to the current SA dropdown and rename it something like VTX Control.   That would also prevent the problems with people trying to get tramp working with SA turned on.

Arti3DPlayer, 2017-08-08:
@hjns @JulianGoesPro 
Will add this in next release:
[image]

Arti3DPlayer, 2017-08-14:
Added in version 0.314.0

closed 2017-08-14 by Arti3DPlayer

JulianGoesPro, 2017-08-14:
love it but there is one problem to that, "OTHER" will be both SmartAudio and TrampTLM... some like the Matek-HV will need the TrampTLM protocol, some like *insert one of those here* will need *SmartAudio*... but true, I doubt we will ever need both at the same time :D 

What about OSD dropdown? :(

Arti3DPlayer, 2017-08-14:
@JulianGoesPro Why you want dropdown for OSD ?:) Why you don't like Switch ?

JulianGoesPro, 2017-08-14:
easy, for the same reason there is a drop down for the vTx, you can connect the OSD to several USARTs :) Not everyone is using it on Tx4, well, not sure what the default is right now...

Arti3DPlayer, 2017-08-14:
But what values should be for OSD ?

Arti3DPlayer, 2017-08-14:
@JulianGoesPro > Matek-HV will need the TrampTLM protocol
oh... show both for other then...

JulianGoesPro, 2017-08-14:
I would remove "Others" and just rename the two "options" to "SmartAudio" and "TrampTLM"

OSD will work for Tx1 Tx2 Tx3 Tx4 Tx6

Arti3DPlayer, 2017-08-14:
@JulianGoesPro We need "Others" option in case that in interview user select Other vtx device. That means "SmartAudio" and "TrampTLM" will be disabled. Maybe just rename to "SmartAudio" and "TrampTLM" and leave "Other" ?

JulianGoesPro, 2017-08-14:
okay, that will do it :)

JulianGoesPro, 2017-08-27:
Awesome job, love it! One little thing, you can't select Tx6 in the OSD USART selection :)