RaceFlightOne/RaceFlightOne #182: Unable to use tones in ESC configuration

yookfpv, 2017-08-23:
-v 0.300
-Hobbywing ESC 4.1 in multishot mode flashed by Raceflight GUI
-Multishot selected.
imperial march or game of throne tôe selected but still have classic tone on startup.

puaka, 2017-08-27:
I flash mine with betaflight for the blheli passthrough, then flash my HW 4 in 1 with 16.65 tones. Calibrate the esc in betaflight(multishot). Then reflash raceflight one, no need to recalibrate esc with raceflight, calibration with rf is giving me weird numbers for PPM min and max, betaflight works better on calibration. Once flashed with 16.65, the tones should work on rf esc menu. Good luck!

PGIII, 2017-09-08:
Thats not going to be calibrated right youre using a lot of power calibrating that way, it should calibrate from 1012 to 1880, around that range, thats correct for race flight. As for it not having the tunes thats likely because we dont have the bin for that esc in there yet we can add it however

proggod, 2017-11-21:
bin should be supported in 403

closed 2017-11-21 by proggod