RaceFlightOne/RaceFlightOne #184: PPM not working with IRC EZUHF receivers (Nano specifically, maybe others)

rdww60, 2017-08-27:
Have been working with adriel and preston to debug without working solution. See private thread between @rdww60, @prestongii, @adriel regarding all the work and effort put in to debug. Symptom is that sticks are not responding when quad is armed (hover and bench) . Plenty of flight logs provided. If no solution will be made, I will ask for refund. Logs from LR9 attached showing that SBUS works, and from CPPM showing that CPPM does not work.
[LR9 Bench.TXT](https://github.com/RaceFlightOne/RaceFlightOne/files/1254654/LR9.Bench.TXT)
[flightlog_2017-71-22_16815_set_telem_tramp_0.txt](https://github.com/RaceFlightOne/RaceFlightOne/files/1254655/flightlog_2017-71-22_16815_set_telem_tramp_0.txt)
[flightlog_2017-71-22_16035 Tramp T Removed.txt](https://github.com/RaceFlightOne/RaceFlightOne/files/1254658/flightlog_2017-71-22_16035.Tramp.T.Removed.txt)
[flightlog_2017-71-19_233926_Bench.txt](https://github.com/RaceFlightOne/RaceFlightOne/files/1254656/flightlog_2017-71-19_233926_Bench.txt)
[flightlog_2017-71-19_17410.txt](https://github.com/RaceFlightOne/RaceFlightOne/files/1254657/flightlog_2017-71-19_17410.txt)

adriel, 2017-08-27:
@rdww60 Just an FYI, think you've got the wrong GitHub name for "adriel", as I'm @adriel and I haven't heard of RaceFlightOne before.

rdww60, 2017-08-27:
Sorry for that, was mentioning their slack name. sorry for the confusion.

proggod, 2018-03-19:
PPM has become very antiquated so I don't think we will be giving much support for this in the future.

closed 2018-03-19 by proggod