RaceFlightOne/RaceFlightOne #187: GUI Enhancement: Live "Rates" Graph

chargerdude70, 2017-09-08:
I looked through items listed here and didn't see this. Be nice to have a live feedback of what rates are set to instead of a static graph. A graph that's tied to the radio data and changes as the sticks do. A person could get a more realistic "feel" for what to expect before going out to field. Especially helpful for a noobie or lesser experienced person that is just getting into changing that aspect of the tuning world.

Arti3DPlayer, 2017-09-08:
Could you explain more detail ?

chargerdude70, 2017-09-08:
An image of a quad that spins in direction of what ever sticks axis you are inputting with transmitter at that time. When rates are set for say 400, 50, 140, quad image rotates 400deg/s but as you move your stick, the image reacts (real time) to the roll rate, roll expo and roll acro plus that your rates set for. I hope I'm conveying this properly. I believe BF has something similar. If anyone else can explain a bit better, please jump in.

Arti3DPlayer, 2017-12-22:
@chargerdude70 http://icdark.github.io/RaceFlightRates/