RaceFlightOne/RaceFlightOne #195: Aux Channel

chargerdude70, 2017-09-19:
Only get 4 channels when detecting RX for Frsky. Used to get 12. Results are I only have 3 AUX channels

PGIII, 2017-09-20:
what version gives you all 12 ? it should be limited to 8 so weird if you used to get more

chargerdude70, 2017-09-22:
Sorry...tile and issue description are misleading. When detecting rx in GUI, the counter tops at 4. It used to go up to 12 just recently and I thought up to 18 in older versions. So its deceiving whether all available channels are getting detected or not.

PGIII, 2017-12-12:
thats not the amount of channels just the number the protocol is, for instance crossfire is the 22nd protocol it detects, doesnt mean it sees 22 channels

closed 2017-12-12 by PGIII