RaceFlightOne/RaceFlightOne #196: Multiple ESC Startup Tones on 343

cr3pitus, 2017-09-20:
Setup - 
Bolt, Revoltv2, RF Motors, Taranis QX7, FrSky XM+ 326 GUI 343 Firmware.

Hardware Has not changed since working setup on 324.

Updated to 326 GUI and 343 Firmware from quopa link in slack channel.

After full setup of quad, plugging in battery when not connected to GUI results in multiple repeated ESC startup tones.

DFU Reflash of 343 yeilds same results.  Not sure if flyable because dont want to cause hardware failure from ESC problem.

Dump included below:
#vr NAME:RaceFlight One;VERSION:0.343.133 BETA;CONFIG:133
#fc HARDWARE:RaceFlight Revolt
#fl size=8960
#fl total=16777216
#me modes active: 0
modes ARMED=5=75=100
modes LEVEL=6=-100=-90
modes ACROLEVEL=6=-10=10
modes FAILSAFE=0=0=0
modes LOGGING=8=-100=-90
modes BUZZER=0=0=0
modes LEDMODE=0=0=0
modes LEDCOLOR=0=0=0
modes DIRECT=0=0=0
modes VTXON=0=0=0
modes BRAINDRAIN=0=0=0
modes PROFILE2=7=-10=10
modes PROFILE3=7=-100=-90
modes QUOPA=0=0=0
modes LEARN=8=-100=10
modes BEEP=0=0=0
throttlecurve1 0=12=25=37=50=62=75=87=100
tpakp1 104=104=100=100=100=100=100=104=104
tpaki1 104=100=95=95=95=95=95=100=104
tpakd1 110=104=100=95=95=95=95=110=110
throttlecurve2 0=12=25=37=50=62=75=87=100
tpakp2 104=104=100=100=100=100=100=104=104
tpaki2 104=100=95=95=95=95=95=100=104
tpakd2 110=104=100=95=95=95=95=110=110
throttlecurve3 0=12=25=37=50=62=75=87=100
tpakp3 104=104=100=100=100=100=100=104=104
tpaki3 104=100=95=95=95=95=95=100=104
tpakd3 110=104=100=95=95=95=95=110=110
set mixer_type=1
set famx=1
set mixer_style=0
set esc_protocol=0
set bit_reverse_esc_1=-1
set bit_reverse_esc_2=-1
set bit_reverse_esc_3=-1
set bit_reverse_esc_4=-1
set bit_reverse_esc_5=-1
set bit_reverse_esc_6=-1
set bit_reverse_esc_7=-1
set bit_reverse_esc_8=-1
set esc_frequency=32000
set idle_percent=5.000
set idle_percent_inv=8.000
set spin_rec_str=750.000
set quopa_style=0
set mout1=0
set mout2=1
set mout3=2
set mout4=3
set mout5=4
set mout6=5
set mout7=6
set mout8=7
set led_count=16
set led_red=10
set led_green=0
set led_blue=0
set led_mode=0
set led_with_usb=0
set telem_smartaudio=0
set telem_sport=0
set telem_spek=0
set telem_msp=0
set telem_rfosd=0
set telem_tramp=0
set telem_crsf=0
set telem_mavlink=0
set adc_current_factor=34.200
set vtx_pitmode_type=0
set vbat_buzzer=1
set vbat_cutoff=3.800
set bat_size=1300
set crsf_otx_cur_hack=1
set gyro_rotation=0
set board_calibrated=1
set man_gy_roll_angle=0.000
set man_gy_pitch_angle=0.000
set man_gy_yaw_angle=0.000
set rf_loop_ctrl=16
set drunk=1
set skunk=2
set pname1=Profile1
set stick_curve1=4
set rc_smoothing1=0.040
set pitch_rate1=375.000
set roll_rate1=475.000
set yaw_rate1=300.000
set pitch_acrop1=0.000
set roll_acrop1=0.000
set yaw_acrop1=0.000
set pitch_expo1=0.000
set roll_expo1=0.000
set yaw_expo1=0.000
set yaw_wc1=0
set roll_wc1=0
set pitch_wc1=0
set yaw_kp1=20.000
set roll_kp1=21.000
set pitch_kp1=22.000
set yaw_ki1=30.000
set roll_ki1=30.000
set pitch_ki1=30.000
set yaw_kd1=50.000
set roll_kd1=23.000
set pitch_kd1=24.000
set kd_limit1=0.350
set ki_limit1=0.150
set slp1=5.000
set sli1=0.100
set sla1=35.000
set sld1=0.030
set res_redux1=0
set yaw_rap1=0.000
set roll_rap1=0.000
set pitch_rap1=0.000
set yaw_ga1=12
set roll_ga1=0
set pitch_ga1=0
set filter_type1=1
set yaw_quick1=70.000
set roll_quick1=45.000
set pitch_quick1=45.000
set yaw_kd_rap1=55.000
set roll_kd_rap1=55.000
set pitch_kd_rap1=55.000
set x_vector_quick1=2.000
set x_vector_rap1=25.000
set y_vector_quick1=2.000
set y_vector_rap1=25.000
set z_vector_quick1=2.000
set z_vector_rap1=25.000
set tpa_kp_curve_type1=1
set tpa_ki_curve_type1=1
set tpa_kd_curve_type1=1
set pname2=Profile2
set stick_curve2=4
set rc_smoothing2=0.040
set pitch_rate2=375.000
set roll_rate2=475.000
set yaw_rate2=300.000
set pitch_acrop2=0.000
set roll_acrop2=0.000
set yaw_acrop2=0.000
set pitch_expo2=0.000
set roll_expo2=0.000
set yaw_expo2=0.000
set yaw_wc2=0
set roll_wc2=0
set pitch_wc2=0
set yaw_kp2=20.000
set roll_kp2=21.000
set pitch_kp2=22.000
set yaw_ki2=30.000
set roll_ki2=30.000
set pitch_ki2=30.000
set yaw_kd2=50.000
set roll_kd2=23.000
set pitch_kd2=24.000
set kd_limit2=0.350
set ki_limit2=0.150
set slp2=5.000
set sli2=0.100
set sla2=35.000
set sld2=0.030
set res_redux2=0
set yaw_rap2=0.000
set roll_rap2=0.000
set pitch_rap2=0.000
set yaw_ga2=12
set roll_ga2=0
set pitch_ga2=0
set filter_type2=1
set yaw_quick2=70.000
set roll_quick2=45.000
set pitch_quick2=45.000
set yaw_kd_rap2=55.000
set roll_kd_rap2=55.000
set pitch_kd_rap2=55.000
set x_vector_quick2=2.000
set x_vector_rap2=25.000
set y_vector_quick2=2.000
set y_vector_rap2=25.000
set z_vector_quick2=2.000
set z_vector_rap2=25.000
set tpa_kp_curve_type2=1
set tpa_ki_curve_type2=1
set tpa_kd_curve_type2=1
set pname3=Profile3
set stick_curve3=4
set rc_smoothing3=0.040
set pitch_rate3=375.000
set roll_rate3=475.000
set yaw_rate3=300.000
set pitch_acrop3=0.000
set roll_acrop3=0.000
set yaw_acrop3=0.000
set pitch_expo3=0.000
set roll_expo3=0.000
set yaw_expo3=0.000
set yaw_wc3=0
set roll_wc3=0
set pitch_wc3=0
set yaw_kp3=20.000
set roll_kp3=21.000
set pitch_kp3=22.000
set yaw_ki3=30.000
set roll_ki3=30.000
set pitch_ki3=30.000
set yaw_kd3=50.000
set roll_kd3=23.000
set pitch_kd3=24.000
set kd_limit3=0.350
set ki_limit3=0.150
set slp3=5.000
set sli3=0.100
set sla3=35.000
set sld3=0.030
set res_redux3=0
set yaw_rap3=0.000
set roll_rap3=0.000
set pitch_rap3=0.000
set yaw_ga3=12
set roll_ga3=0
set pitch_ga3=0
set filter_type3=1
set yaw_quick3=70.000
set roll_quick3=45.000
set pitch_quick3=45.000
set yaw_kd_rap3=55.000
set roll_kd_rap3=55.000
set pitch_kd_rap3=55.000
set x_vector_quick3=2.000
set x_vector_rap3=25.000
set y_vector_quick3=2.000
set y_vector_rap3=25.000
set z_vector_quick3=2.000
set z_vector_rap3=25.000
set tpa_kp_curve_type3=1
set tpa_ki_curve_type3=1
set tpa_kd_curve_type3=1
set rx_protocol=4
set rx_usart=0
set arm_method=1
set rx_inv_direction=0
set pitch_deadband=0.015
set roll_deadband=0.015
set yaw_deadband=0.015
set throttle_deadband=0.000
set aux1_deadband=0.000
set aux2_deadband=0.000
set aux3_deadband=0.000
set aux4_deadband=0.000
set pitch_midrc=992
set roll_midrc=992
set yaw_midrc=985
set throttle_midrc=993
set aux1_midrc=987
set aux2_midrc=987
set aux3_midrc=987
set aux4_midrc=987
set pitch_minrc=189
set roll_minrc=192
set yaw_minrc=189
set throttle_minrc=193
set aux1_minrc=189
set aux2_minrc=189
set aux3_minrc=189
set aux4_minrc=189
set pitch_maxrc=1794
set roll_maxrc=1787
set yaw_maxrc=1790
set throttle_maxrc=1794
set aux1_maxrc=1790
set aux2_maxrc=1790
set aux3_maxrc=1790
set aux4_maxrc=1790
set pitch_map=2
set roll_map=1
set yaw_map=3
set throttle_map=0
set aux1_map=4
set aux2_map=5
set aux3_map=6
set aux4_map=7
set aux5_map=8
set aux6_map=9
set aux7_map=10
set aux8_map=11
set aux9_map=12
set aux10_map=13
set aux11_map=14
set aux12_map=15
set rc_calibrated=1
set bind=0
set short_throw=1
set omega0=1.000
set omega1_yaw=30.000
set omega1_roll=40.000
set omega1_pitch=40.000
set omega2_yaw=0.088
set omega2_roll=0.088
set omega2_pitch=0.088

cr3pitus, 2017-09-20:
as an update going to 340 resulted in same issue, going back to 324 resolved issue - so source of problem is likely in some change between 324 and 340

PGIII, 2017-09-20:
Thats just the new auto tuning stuff taking longer to start up, its not a big deal and is normal for this version, we can look into not making it do that though, what startup tone are you using ?

cr3pitus, 2017-09-21:
Hey thanks pg3 for responding.  Default startup tones.  I have run other times in the past like the mario tone.  Some people have a bolt revolt setup and haven't had it happen so maybe if you've never changed tones it's less likely to occur.

Mio44, 2017-09-27:
It also happens in 349, using the default tone (tadada-ta-da). When i plug in lipo, it goes: tadada-tadada-ta-da.

Matthias7Song, 2017-10-14:
Also happens in 361.

PGIII, 2018-03-19:
fixed in latest

closed 2018-03-19 by PGIII