RaceFlightOne/RaceFlightOne #199: [Hardware Support] Add smart audio support for other VTXes than Unify

necloveksk, 2017-09-27:
I noticed that only unify works with smart audio. For example the EWRF e7086TM3 VTX does support smart audio (works on BF) but RF doesnt recognize it and cant control it.

PGIII, 2017-09-27:
Sounds like they didnt clone the protocol properly, if they want it supported theyll need to send us a vtx to test with, both Immersion and TBS have worked with us and gotten us vtx's so that we can make sure they work.

Mio44, 2017-09-27:
What about Tramp v1? Is the issue there the same? Is it some glitch they have fixed in Tramp v2?

supiiik, 2017-09-28:
[EWFR.zip](https://github.com/RaceFlightOne/RaceFlightOne/files/1340227/EWFR.zip)
Hi, I have done a few measurements with logic analyzer and I can see, that EWRF is using serial communication with 1 stop bit 8N1. That can be the issue in this case. Betaflight is also communicating with 8N1. 
When I connect TBS unify, that is using 8N2 (or 8N4 by the measurements actually...), the communication is working fine even when the BF is still sending packets with 8N1.
logicdata with EWRF communication are in attachment

proggod, 2018-03-19:
planned for v1.1

closed 2018-11-29 by PGIII