RaceFlightOne/RaceFlightOne #200: Revolt V2 (Skitzo) unresponsive after trying quopa mode.

Dumptruck100, 2017-09-29:
I do have another issue. I have a revolt skitzo I flashed to 320 or 326. Flys great until I try quopa mode, once I do that, quad wont re arm, and the blue led on the fc is off. Only way to get to work again is to unplug and reconnect battery. Watched it closely on the bench. Blue status led ok until qoupa switch is turned on, then blue led starts to flicker then goes out and quad is unresponsive.
 I did notice that 2 of the esc's are missing a tiny resistor, got them stuffed into a Hiro 180x frame so like no room for 30a esc's, so I was thinking that might be the problem when esc's try to reverse.

breakthroughsecurity, 2017-10-18:
I have this same issue.  Blue led acts the same as well.  All of my escs are deffinitely not damaged.  I can reverse the direction of the props in the configurator so i assume the escs should be able to be reversed and this should work.

I have configured quopa mode to a switch on my taranis. To test i fly and end up upside down then I disarm.  Then i hit the quopa mode switch and leave it engaged.  At this point i hear the escs make some noise (it sounds exactly like all of the quopa videos on youtube).  I then try to arm but it never arms and tries to flip back over.  What could cause it to not rearm and try to flip over.   I have tried using frimware 320 and 343 and get the same results. I am running these escs https://www.amazon.com/gp/product/B06XBWSSCF/ref=oh_aui_detailpage_o00_s00?ie=UTF8&psc=1  and this fc https://www.getfpv.com/lumenier-skitzo-flight-controller-powered-by-raceflight.html if that helps.

proggod, 2017-11-21:
try 402, should be fixed

closed 2017-11-21 by proggod