RaceFlightOne/RaceFlightOne #201: Configurator very buggy on OSX High Sierra

denovich, 2017-10-01:
I had no issues with the configurator for some months... but after updating OSX to High Sierra, I get numerous timeouts and errors saving/loading data. 

Below is what I noticed in the log files...

[image]

[image]

denovich, 2017-10-01:
This happened on versions 363, 326

Arti3DPlayer, 2017-10-01:
@denovich Weird, I have mac os sierra 10.12.6 and it is fine.

denovich, 2017-10-01:
I had practically zero issues prior to the upgrade to OSX 10.13.  
The problems must be low level... running RF1C in a Windows VM does not work around the issue.

denovich, 2017-10-01:
Doh!  Seems to be related to one particular Revolt. I tested on my older MBP which is running El Cap.  The first time with a different Revolt... it wasn't until later that I tried the problematic one on the older laptop.

So false alarm.

closed 2017-10-01 by denovich