RaceFlightOne/RaceFlightOne #210: FrSky telemetry freezes shortly after takeoff- 320 and several Beta builds

kenkamm, 2017-10-21:
FrSky telemetry doesn’t work reliably. Tested on XSR and R-XSR, both of which running latest FrSky firmware. (Works fine on Betaflight) Telemetry works at first, but after taking to the air the numbers freeze and updates to the Taranis stop. Thus, any voltage or mAh warnings set up in the Taranis don’t work and you end up over-discharging your batteries.

djfriesema, 2017-10-21:
I am having the same issue, with repeated "Sensor Loss" alerts from the Taranis.

2mod, 2017-10-22:
tried the x4r too?

djfriesema, 2017-10-23:
Is this only an issue with the XSR?  If so I'll get an X4R and try it.

2mod, 2017-10-23:
give it a go, the more detail we can get, the easier its for the devs to fix

djfriesema, 2017-10-23:
One more thing, I'm getting this on 320 which is supposed to be a stable build, so it's not just beta builds where this is occuring.

2mod, 2017-10-23:
the most important is the most recent build
ie if it doesnt work on build A but works on B, you could consider it having been fixed (assuming its consistent ofc)
since i doubt there will be separate bug fixing for older builds, as that would exponentially increase the maintenance and risk of further bugs

GateslayerFPV, 2017-11-15:
I think I'm getting this as well... after arming... you get 16.8v as a readout... when you disarm you get actual voltage.

proggod, 2017-11-21:
can you guys test on 402

proggod, 2017-11-21:
reopen if still broken

closed 2017-11-21 by proggod