RaceFlightOne/RaceFlightOne #211: Spektrum TLM broken in build 374 (quad twitches at idle, eventually failsafes?)

denovich, 2017-10-22:
If Spektrum TLM (soldered to TX1) is enabled, upon arm, quad twitches randomly for a few seconds before the motors stop (failsafe?). The quad can be re-armed, but the same behavior happens.

If Spektrum TLM is then disabled via the Configurator, the problem goes away. I can arm and fly as usual.

JulianGoesPro, 2017-10-22:
that is how it looks for me^^

[image]

RWooten68, 2017-10-24:
Some has reported putting the ser on rx3 fixed the issue.

JulianGoesPro, 2017-10-28:
@RWooten68 will look into that, thanks!

RWooten68, 2017-10-28:
@JulianGoesPro Don't that wasn't the same issues.

JulianGoesPro, 2017-10-28:
@RWooten68 well, it makes sense that that exact problem goes away when using Rx3, since it will disable SpekTLM which is the reason for this problem :)

denovich, 2017-12-04:
I think this is now closed.  TLM seems to work on 419 at least.

closed 2017-12-12 by PGIII