RaceFlightOne/RaceFlightOne #213: it does not want to save when I change to profile 2

robomaniac, 2017-10-22:
Version: 0.320.1
FC Version: 0.320.125 BETA
---------------------------------------
I switch to profile 2 when I am in "_PID tuning tab_" and can not save.

I am experimenting with rates, PID, filters, etc so I want to select between profile 1, 2 and 3 but when I finish selecting profile 2, for example, I can not save even though it show all the new values.

Arti3DPlayer, 2017-10-23:
I checked all is fine. Is this happens every time for you ?

robomaniac, 2017-10-23:
I only tried once on one build. Let me try again soon with my other build
and will make video.

robomaniac, 2017-10-25:
please have a look at this private video
https://youtu.be/sV0mArmj5vc

Arti3DPlayer, 2017-10-25:
@robomaniac this is not issue. RF board has 3 different profiles. Each profile has own values.

For example you change rc_smoothing on profile1, then you can go on profile two and you will see another value, it is different variables.

closed 2017-10-25 by Arti3DPlayer

robomaniac, 2017-10-25:
I don't think we understand each other.

The goal for 3 profiles to to switch between them and go fly. Here is example Profile 1 is my "perfect" setup but I want to play with new filter or PIDs, etc. without loosing those "perfect" values, I create Profile 2 to experiment but I can't save those profile to my FC. I am in profile 2 with new values but there is no way to send them to FC.

Is that clear? Just want to make sure we understand each other.

Or what is the point of having multiple profile?

Arti3DPlayer, 2017-10-25:
@robomaniac oh, now I understand. When you change value on video and click save you value should be saved. Isn't it ?

Arti3DPlayer, 2017-10-25:
>I create Profile 2 to experiment but I can't save those profile to my FC. I am in profile 2 with new values but there is no way to send them to FC.

On your firmware you have:
rc_smoothing1 = 50
rc_smoothing2 = 50
rc_smoothing3 = 50

On video you change value on profile3 to 30 for example, so

rc_smoothing1 = 50
rc_smoothing2 = 50
rc_smoothing3 = 30

Arti3DPlayer, 2017-10-25:
But profiles on firmware you should change via your RADIO

robomaniac, 2017-10-25:
yes, the goal of the profile 1,2 and 3 is that you have a switch on the radio to select between them.

if you would "save" profile 2 to FC it would not matter because when you boot quad and turn on remote and the switch is set at profile 1, it will select profile 1 and not what you have set in GUI. The switch overrides.

ok I understand.

I will set a switch then. The only thing why I did not want to have switch is because I am scared that during race day my switch is set to profile 3 and I am suppose to fly on profile 1. But then again, I will make sure profile 1 is always the best settings and use profile 3 to experiment. So when I turn on Taranis I place switch all "up" so by default set to profile 1. And switch is better since I can compare between the 2 really fast.
 I understand myself :)

I guest you can close this. thanks again for your help!